hhk7734/python3-gpiod #13: Question: is this the same as `python3-libgpiod` package in Debian

BrendanSimon, 2020-09-28:
Hi,

Is this the same as `python3-libgpiod` package in Debian?

If not, then what is the advantage in using this python module (python3-gpiod)?

Is there a more appropriate forum to discuss questions (instead of github issues?)

Thanks, Brendan.

hhk7734, 2020-09-28:
No advantage.

Until libgpiod v1.4.0 was released, except for the raspberry pi, there were not many boards where the python3-libgpiod package was provided through the package manager. So, I have created this package on PyPI based on the libgpiodcxx package which is familiar to me.

BrendanSimon, 2020-09-28:
Ok.  Thanks for the update.  I have the official python3-libgpiod working on my board so I'll stick with that.  If I have any issues I will come back and check if your python module.

JohanAR, 2021-03-17:
I started using python3-libgpiod from raspbian (1.2-3+rpi1) and it seems like the classes and methods has changed a lot since then. So wouldn't there be an advantage in using the latest version from pip in that you won't have to rewrite your code when the deb is updated?

clach04, 2023-02-26:
> If not, then what is the advantage in using this python module (python3-gpiod)?

From my perspective the advantage of this project is that as a Python developer I can see more information in the stack trace.

I'm also more optimistic about getting updates as I can potentially patch it myself ;-)

BrendanSimon, 2023-02-28:
Sounds like some reasonable arguments, especially since Debian packages are generally so far behind when there is a new release, and that only critical updates are available.  So using python packages from PyPi makes it easier to get new features and bug fixes for my python apps and still use a stable operating system.
Cheers !!

closed 2023-02-28 by BrendanSimon