hhk7734/python3-gpiod #15: Question: how to release a previously requested line?

w4tsn, 2021-01-11:
Hey there: I've a case where a framework first initializes the gpio line as input and then as interrupt. I don't know this in the input initialization so I'm first making a line request as type input, but then I would like to request the line as type event / interrupt. How can I release the previously set line or is there a way to change the request type?

I've tried `gpiod.libgpiod.gpiod_line_release()`, which does not throw an error, but it later complains that the line is still busy. Any idea what I'm doing wrong?

hhk7734, 2021-01-11:
There is no function in `gpiod.libgpiod` that throws the error.

Why initialize to input first? line_request.EVENT_XX already means input.

If you want to request after `line.request(config)`, try after `line.release()`.

w4tsn, 2021-01-11:
Ah, well my bad. I didn't find that function in the code. I've used `.release()` now as you suggested and that works perfectly. Thanks!

The reason I've to use this for now (the quick+dirty solution) is that the software / framework I work with separates input/output and interrupt through two abstract methods that both get called in many different configuration. So I have to initialize the line as input in the one function and release+request it as interrupt again in the next because in the context of the input-function I don't know if I'm actually dealing with an interrupt config or just a regular input.

closed 2021-01-11 by w4tsn