hhk7734/python3-gpiod #16: fraction of a second doubled on event timestamp returned?

JimTheKiwi, 2021-03-27:
https://github.com/hhk7734/python3-gpiod/blob/master/py_src/gpiod/libgpiod/__init__.py#L1053
`    sec = evdata.timestamp / 1_000_000_000`

Shouldn't that division be integer // not float / ?
I suspect that sec is a floating point number still including the fraction of a second, and that fraction is then being added into event.ts twice via both seconds and microseconds:
```
        seconds=sec % 86400,
        microseconds=(evdata.timestamp % 1_000_000_000) // 1000,
```
I am seeing timestamps with the fraction of a second doubled, eg 1616864113.8 should be about 1616864113.4 and  1616864114.6 should be about 1616864113.8 (ie overflows to the next second).  Timestamps being returned at 25Hz rise by 2 seconds in a 1 second period, then fall back 1 second and rise again.
Platform NVIDIA Jetson L4T 4.5 (Ubuntu 18.04 on ARM)

hhk7734, 2021-03-29:
Would you make a PR?

JimTheKiwi, 2021-03-29:
A PR to add one "/" would be pretty quick to do, but I'm reluctant to send untested code and now I am not set up to test with GPIO on my Dev system.

The test system I used earlier is not with me now (we migrated it from an ubuntu binary packaged python3-gpiod to a small C++ client calling libgpiod directly).  It was only later I guessed where the timing error might be caused, looked at this repo and raised my unproven theory to try to save others some time.

It occured to me that since the seconds=sec%86400 appears to be successfully passing a float including the fraction of a second, then you could decide leave the division as "/" floating and instead drop the "microseconds=" line.  It might be nanosecond precise instead of truncating to microseconds.  That is a bit harder to understand and might run into precision problems if it uses too small a float somewhere.

hhk7734, 2021-03-29:
```python
>>> datetime.timedelta(days=100, seconds=100.5, microseconds=100)
datetime.timedelta(100, 100, 500100)
```

This is clearly unintended behavior.
I'll test it and fix it. Thanks :)

JimTheKiwi, 2021-03-29:
It could be intentional to support seconds=100.5, but it does seem unusual to then also include microseconds and add the two together.

Thanks for your work on this, good luck.

FYI You probably won't run into this headache, but some devices use a semi-rounded time with a slightly negative fraction of a second added in for strict compatibility with legacy devices that _round_ to poorer precision.  Explanation from the [manual of a precision timing receiver](https://www.u-blox.com/sites/default/files/ZED-F9T-00B_IntegrationManual_%28UBX-19005590%29.pdf) section 3.8.6 UTC Representation:

> UTC ... is normally reported with two decimal places.
> ... date and time are separated into seven distinct integer fields... year, month, day, hour, min and sec have fairly obvious meanings...
> The seventh field is called nano and it contains the number of nanoseconds by which the rest of the time and date fields need to be corrected to get the precise time. So, for example, the UTC time 12:49:23.521 would be reported as: hour: 12, min: 49, sec: 23, nano: 521000000.
> It is however important to note that the first six fields are the result of rounding to the nearest hundredth of a second. Consequently the _nano value can range from -5000000 (i.e. -5 ms) to +994999999 (i.e. nearly 995 ms)._

commit a1cdc8a1 (2021-04-13): libgpiod: fix issue where microseconds is calculated twice

To check event time, this lib uses `datetime`. Using `timedelta` with
float type `seconds` parameter causes the problem of adding the
fractional part to the `microseconds`.

Thanks @JimTheKiwi

close #16
closed 2021-04-13 by hhk7734