hhk7734/python3-gpiod #28: Problem on Odyssey x86j4125 SBC

bwmtx, 2022-05-21:
I'm trying to get mqtt-io (https://github.com/flyte/mqtt-io) working with this driver, on Debian Bullseye.

The Odyssey SBC has a different pin number schema.  See here --> https://wiki.seeedstudio.com/ODYSSEY-X86J4105-GPIO/

The error I get is...

```
python3 -m mqtt_io config.yaml 
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/__main__.py", line 85, in <module>
    main()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/__main__.py", line 80, in main
    mqtt_gpio.run()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 1227, in run
    self._init_digital_inputs()
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/server.py", line 387, in _init_digital_inputs
    gpio_module.setup_pin_internal(PinDirection.INPUT, in_conf)
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/modules/gpio/__init__.py", line 207, in setup_pin_internal
    return self.setup_pin(
  File "/usr/local/lib/python3.9/dist-packages/mqtt_io/modules/gpio/gpiod.py", line 69, in setup_pin
    line: "gpiod.line" = self.chip.get_line(pin)
  File "/usr/local/lib/python3.9/dist-packages/gpiod/libgpiodcxx/__init__.py", line 205, in get_line
    raise IndexError("line offset out of range")
IndexError: line offset out of range
```

Is there a way to make this work?

Thanks

hhk7734, 2022-05-22:
Maybe remove this package. Then try again after `sudo apt install python3-libgpiod`

bwmtx, 2022-05-22:
@hhk7734 

Will give it a try.  Thank you

bwmtx, 2022-05-22:
@hhk7734 

No joy.  I've tried all pin number options.  Gpio, board and what odyssey docs say.  Config file will let me use gpio pin 4 and board pin 7 without error.  When using board pin 7 the log output changes True but connecting to 3.3v or ground doesn't change to false.  When I enter odyssey pin 337 I get the error 'IndexError: line offset out of range'.  My thought is the board number 7 needs to be mapped to the odyssey number of 337.  Not entirely sure though.

hhk7734, 2022-05-22:
If you installed libgpiod, try `gpioinfo` and check which gpiochip the board has and what the range of gpioline is.

bwmtx, 2022-05-22:
```
gpiochip0 - 80 lines:
	line   0:      unnamed       unused   input  active-high 
	line   1:      unnamed       unused   input  active-high 
	line   2:      unnamed       unused   input  active-high 
	line   3:      unnamed       unused   input  active-high 
	line   4:      unnamed       unused   input  active-high 
	line   5:      unnamed       unused   input  active-high 
	line   6:      unnamed       unused   input  active-high 
	line   7:      unnamed       unused   input  active-high 
	line   8:      unnamed       unused   input  active-high 
	line   9:      unnamed       unused   input  active-high 
	line  10:      unnamed       unused  output  active-high 
	line  11:      unnamed       unused  output  active-high 
	line  12:      unnamed       unused  output  active-high 
	line  13:      unnamed       unused   input  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
	line  16:      unnamed       unused   input  active-high 
	line  17:      unnamed       unused   input  active-high 
	line  18:      unnamed       unused   input  active-high 
	line  19:      unnamed       unused   input  active-high 
	line  20:      unnamed       unused   input  active-high 
	line  21:      unnamed       unused   input  active-high 
	line  22:      unnamed       unused   input  active-high 
	line  23:      unnamed       unused   input  active-high 
	line  24:      unnamed       unused   input  active-high 
	line  25:      unnamed       unused   input  active-high 
	line  26:      unnamed       unused   input  active-high 
	line  27:      unnamed       unused  output  active-high 
	line  28:      unnamed       unused  output  active-high 
	line  29:      unnamed       unused   input  active-high 
	line  30:      unnamed       unused   input  active-high 
	line  31:      unnamed       unused   input  active-high 
	line  32:      unnamed       unused   input  active-high 
	line  33:      unnamed       unused   input  active-high 
	line  34:      unnamed       unused   input  active-high 
	line  35:      unnamed       unused   input  active-high 
	line  36:      unnamed       unused   input  active-high 
	line  37:      unnamed       unused   input  active-high 
	line  38:      unnamed       unused   input  active-high 
	line  39:      unnamed       unused   input  active-high 
	line  40:      unnamed       unused   input  active-high 
	line  41:      unnamed       unused   input  active-high 
	line  42:      unnamed       unused  output  active-high 
	line  43:      unnamed       unused   input  active-high 
	line  44:      unnamed       unused   input  active-high 
	line  45:      unnamed       unused   input  active-high 
	line  46:      unnamed       unused   input  active-high 
	line  47:      unnamed       unused   input  active-high 
	line  48:      unnamed       unused   input  active-high 
	line  49:      unnamed       unused   input  active-high 
	line  50:      unnamed       unused   input  active-high 
	line  51:      unnamed       unused   input  active-high 
	line  52:      unnamed       unused   input  active-high 
	line  53:      unnamed       unused   input  active-high 
	line  54:      unnamed       unused   input  active-high 
	line  55:      unnamed       unused   input  active-high 
	line  56:      unnamed       unused   input  active-high 
	line  57:      unnamed       unused   input  active-high 
	line  58:      unnamed       unused   input  active-high 
	line  59:      unnamed       unused   input  active-high 
	line  60:      unnamed       unused  output  active-high 
	line  61:      unnamed       unused  output  active-high 
	line  62:      unnamed       unused   input  active-high 
	line  63:      unnamed       unused   input  active-high 
	line  64:      unnamed       unused   input  active-high 
	line  65:      unnamed       unused   input  active-high 
	line  66:      unnamed       unused   input  active-high 
	line  67:      unnamed       unused  output  active-high 
[... 126 lines cut ...]
	line  11:      unnamed       unused   input  active-high 
	line  12:      unnamed       unused   input  active-high 
	line  13:      unnamed       unused   input  active-high 
	line  14:      unnamed       unused   input  active-high 
	line  15:      unnamed       unused   input  active-high 
	line  16:      unnamed       unused   input  active-high 
	line  17:      unnamed       unused   input  active-high 
	line  18:      unnamed       unused   input  active-high 
	line  19:      unnamed       unused   input  active-high 
	line  20:      unnamed       unused   input  active-high 
	line  21:      unnamed       unused   input  active-high 
	line  22:      unnamed       unused   input  active-high 
	line  23:      unnamed       unused   input  active-high 
	line  24:      unnamed       unused   input  active-high 
	line  25:      unnamed       unused   input  active-high 
	line  26:      unnamed       unused   input  active-high 
	line  27:      unnamed       unused   input  active-high 
	line  28:      unnamed       unused   input  active-high 
	line  29:      unnamed       unused   input  active-high 
	line  30:      unnamed       unused   input  active-high 
	line  31:      unnamed       unused   input  active-high 
	line  32:      unnamed       unused   input  active-high 
	line  33:      unnamed       unused   input  active-high 
	line  34:      unnamed       unused   input  active-high 
```

bwmtx, 2022-05-22:
I think gpiochip3 is the one attached to the 40 pin raspberry pi compatible header.

hhk7734, 2022-05-23:
Unless you look up this board's gpio mapping in the kernel code, turning all lines on and off sequentially might be the quickest way to find the correct line.

closed 2022-08-16 by hhk7734