hhk7734/python3-gpiod #31: OSError: [Errno 16] Device or resource busy

clach04, 2023-02-26:
I'm having some issues with permissions when opening a pin up for write. What's weird is this works fine with RPi.GPIO. Is there the concept of shared access? and/or is there an equivalent in this library to:

            RPi.GPIO.setmode(rpi_gpio.BCM)
            RPi.GPIO.setwarnings(False)

I don't know much about GPIO/SPI so this may be a user error ;-)

I have an eink display that uses GPIO and SPI. I wrote a Python "driver" based on some C code for a Raspberry Pi, and it works (as well as the C code). Relying on spidev and RPi.GPIO.

I ported the code over to both this library and the OS one from debian/raspbian (I suspect it's https://git.kernel.org/pub/scm/libs/libgpiod/libgpiod.git/) and get conflicts opening the same pin SPI is using:

$ cat alt_cs_issue.py

    import gpiod  # expected to be https://github.com/hhk7734/python3-gpiod
    print(dir(gpiod))
    print(gpiod.__file__)

    CS_PIN = 8
    chip = gpiod.chip('gpiochip0')
    """
    $ gpioinfo |grep 8
            line   8:      unnamed   "spi0 CS0"  output   active-low [used]

    """
    _CS_PIN = chip.get_line(CS_PIN)
    config = gpiod.line_request()
    config.consumer = __file__
    config.request_type = gpiod.line_request.DIRECTION_OUTPUT

    _CS_PIN.request(config=config)

output.

      (py3venv) pi@pizero:~/c/eink_ssd1619_er-epm042-1/ER-EPM042-1_Raspberry_Pi/ER-EPM042A1-1R/python $ pip list |grep gpio
    gpiod         1.5.3
    (py3venv) pi@pizero:~/c/eink_ssd1619_er-epm042-1/ER-EPM042-1_Raspberry_Pi/ER-EPM042A1-1R/python $ gpioinfo |grep 'line   8'
            line   8:      unnamed   "spi0 CS0"  output   active-low [used]
    (py3venv) pi@pizero:~/c/eink_ssd1619_er-epm042-1/ER-EPM042-1_Raspberry_Pi/ER-EPM042A1-1R/python $ python3 alt_cs_issue.py
    ['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', 'chip', 'chip_iter', 'find_line', 'kernel', 'libgpiod', 'libgpiodcxx', 'line', 'line_bulk', 'line_event', 'line_iter', 'line_request', 'make_chip_iter']
    /home/pi/c/eink_ssd1619_er-epm042-1/ER-EPM042-1_Raspberry_Pi/ER-EPM042A1-1R/python/py3venv/lib/python3.7/site-packages/gpiod/__init__.py
    Traceback (most recent call last):
      File "alt_cs_issue.py", line 17, in <module>
        _CS_PIN.request(config=config)
      File "/home/pi/c/eink_ssd1619_er-epm042-1/ER-EPM042-1_Raspberry_Pi/ER-EPM042A1-1R/python/py3venv/lib/python3.7/site-packages/gpiod/libgpiodcxx/__init__.py", line 572, in request
        rv = libgpiod.gpiod_line_request(_m_line, conf, default_val)
      File "/home/pi/c/eink_ssd1619_er-epm042-1/ER-EPM042-1_Raspberry_Pi/ER-EPM042A1-1R/python/py3venv/lib/python3.7/site-packages/gpiod/libgpiod/__init__.py", line 499, in gpiod_line_request
        return gpiod_line_request_bulk(bulk, config, [default_val])
      File "/home/pi/c/eink_ssd1619_er-epm042-1/ER-EPM042-1_Raspberry_Pi/ER-EPM042A1-1R/python/py3venv/lib/python3.7/site-packages/gpiod/libgpiod/__init__.py", line 542, in gpiod_line_request_bulk
        return _line_request_values(bulk, config, default_vals)
      File "/home/pi/c/eink_ssd1619_er-epm042-1/ER-EPM042-1_Raspberry_Pi/ER-EPM042A1-1R/python/py3venv/lib/python3.7/site-packages/gpiod/libgpiod/__init__.py", line 403, in _line_request_values
        status = ioctl(fd, GPIO_GET_LINEHANDLE_IOCTL, req)
    OSError: [Errno 16] Device or resource busy

vfazio, 2023-02-28:
does `gpioinfo` show the pin as already claimed by another owner?

```
gpiochip0 - 54 lines:
	line   0: "rpi0_id_eeprom_sda" unused input active-high 
	line   1: "rpi0_id_eeprom_scl" unused input active-high 
```
You should see "unused" in order for the claim to work. Oftentimes, other RPi GPIO libraries bypass the libgpio claim system and use direct memory writes (see libbcm2835). This is obviously problematic because something could hit the GPIO without being blocked by the libgpio claim system.

clach04, 2023-03-02:
Thanks for taking the time to reply, it's definitely marked as `[used]`, see output above.

You've given me the confidence that the original source I was porting is hacky and that I need to go back to the spi code and reimplement that first.

Again, thank you!

closed 2023-03-02 by clach04